dcwbrown/dwire-debug #2: Can't seem to get this to work, looking for some advice

doubleodoug, 2016-08-23:
I'm working with an ft232r and an ATtiny85 fused to E:FE, H:9F, L:62 (8MHz oscillator divided by 8, debugWIRE enabled, SPI programming enable). I can't seem to get dwire-debug to do much of anything except lock up the CPU. I'm using archlinux with gcc 6.1.1 20160802.

Was this developed using an ATtiny fused for a 16MHz CPU instead of my 1MHz? Does this work only on the ATtiny45 and not ATtiny85? I don't have my own "official" debugwire programmer so my ability to debug/develop this on my own is limited.

Any advice you have would be greatly appreciated, I'd really like to get debugwire working.

dcwbrown, 2016-08-23:
My fuses were E:FF H:9F L:62. As far as I remember this is as they come
from Atmel but with debugwire enabled. 

I think the clock setting should be correct. (It is the 3rd line of
GlobalData.c if you do need to change it.) 

The only difference I see is that I had self programming disabled. While
I don't think it should matter, you could try changing that. 

What happens when you run dwdebug? 

It should say 'unconnected' to which you type 'reset' and it will look
through the available com ports. Mine looks like this: 

 $ ./dwdebug.exe
 Unconnected. > reset
 SerialRead expected 1 bytes, received 0 bytes from COM1
 ATtiny45 found on COM3
 0000: c00e rjmp 000f (+15) >

A thought - is there a support package you need to install for the
ft232r? On Linux, the serial port is initialised in lines 40 to 55 of
SerialPort.c. These set the BaudRate. It is possible that the standard
 OS handler cannot work with non-standard baud rates (which this is),
and that you need to install
 an ft232r specific driver. 

Good luck -- Cheers -- Dave. 

 

Links:
------
[1] https://github.com/dcwbrown/dwire-debug/issues/2
[2]
https://github.com/

dcwbrown, 2016-08-24:
FYI have ordered a couple of ATTiny85s and will test probably tomorrow.
Another thought - have you tried an erased ATTiny85? It is possible for a bug in your code to block debugwire - I don't remember exactly how I managed that but it may have been something like putting it into low power mode in an infinite loop.

dcwbrown, 2016-08-25:
I'm pleased to report that a clean ATTiny85 with the debug wire enable flag programmed is working nicely with dwdebug - it is recognised correctly and dwdebug honours its flash and memory sizes.

doubleodoug, 2016-08-26:
My wiring is exactly the same as yours (except I used a schottky diode instead of a normal one). If I disconnect the ATtiny but leave the diode between Tx and Rx I get back every byte I send, so I know the hardware is working...

I get this on Windows 10 x64: http://i.imgur.com/rxT5vmZ.png
And this on Archlinux: http://i.imgur.com/JFHqamk.png
I also tried this on my Raspberry Pi running raspbian as well, same results as archlinux.

So basically on Windows it opens the COM port but doesn't get any data back and on ArchLinux it does... nothing. I get the same result if the ATtiny is not connected. I get the same result if I run dwdebug and type reset.

The program I have running on the ATtiny toggles PB4 just for some visual confirmation the CPU is running (no interrupts, just a while loop and a delay). When I run dwdebug the LED stops blinking (presumably meaning the CPU is stopped), but I get this same result if I just temporarily short RESET to ground so all this really means is that the FT232R is at least sending something.

There are two FTDI packages available in archlinux, libftdi and libftdi-compat. Whether they are both installed or not installed makes no difference. The loaded modules that seem relevant are: http://i.imgur.com/jDVmlYF.png

I've used this FTDI chip many times and it's always worked without issue.

I'm not sure what else I can try. Could you send me a compiled .exe to test? Also thanks so much for the quick and complete response, I really appreciate it.

dcwbrown, 2016-08-26:
How disappointing :-(.
When I run on Windows 10 x64 I get:
```
$ make
i686-w64-mingw32-gcc -std=gnu99 -Wall -o dwdebug.exe -Dwindows dwdebug.c -lKernel32 -lComdlg32
#i686-w64-mingw32-gcc -g -oo -std=gnu99 -Wall -o dwdebug.exe -Dwindows dwdebug.c -lKernel32 -lComdlg32
ls -lap dwdebug.exe
-rwxr-xr-x 1 dave None 159872 Aug 26 10:30 dwdebug.exe
./dwdebug.exe f0,q
SerialRead expected 1 bytes, received 0 bytes from COM1
ATtiny85 found on COM3
0000:   ff ff ff ff  ff ff ff ff   ff ff ff ff  ff ff ff ff   ................
0010:   ff ff ff ff  ff ff ff ff   ff ff ff ff  ff ff ff ff   ................
0020:   ff ff ff ff  ff ff ff ff   ff ff ff ff  ff ff ff ff   ................
0030:   ff ff ff ff  ff ff ff ff   ff ff ff ff  ff ff ff ff   ................
0040:   ff ff ff ff  ff ff ff ff   ff ff ff ff  ff ff ff ff   ................
0050:   ff ff ff ff  ff ff ff ff   ff ff ff ff  ff ff ff ff   ................
0060:   ff ff ff ff  ff ff ff ff   ff ff ff ff  ff ff ff ff   ................
0070:   ff ff ff ff  ff ff ff ff   ff ff ff ff  ff ff ff ff   ................
```
I notice your dwdebug binary is slightly larger. 

Here's the binary generated above (.txt added to bypass github restrictions.)
[dwdebug.exe.txt](https://github.com/dcwbrown/dwire-debug/files/438965/dwdebug.exe.txt)

I'm currently using just a cheap 1N914/1N4148 without problem. 

I'll try a schottky and see if it gives me any problems - what type are you using? I have SC07443 and SR304 to hand.

I'll also update my cygwin install and see if that creates a binary matching the size of yours.
-- Dave.

doubleodoug, 2016-08-26:
Still doesn't work with that binary... It's got to be my diode then right? I'm not sure why it would matter but everything else is the same...
I'm using one of these: http://www.digikey.com/product-detail/en/diodes-incorporated/1N5817-T/1N5817DICT-ND/190530

I'll pick up a 1N4148 and report back. Thanks so much for the help.

dcwbrown, 2016-08-26:
I've updated cygwin and am now getting the same size binary as you, and it still works.
I've made up a wire with a schottky and it still works.
Have tried 3 other FT232RL boards, and it still works. (Some of them have a different pinout from my picture.)

I suppose it could just possibly be a fake FT232 that doesn't support non-standard speeds. It would likely have worked just fine for other uses. 

Have you tried without the LED connected? - it could I suppose be glitching the power on the FT232.

Good luck -- Dave.

doubleodoug, 2016-08-27:
I got some 1N914 diodes... still nothing...
Tried it without the LED, tried it at 5/3.3/1.8 volts... still nothing...
Tried a different ATtiny I found in my parts bin (ATtiny25 I've just fused to E:FF H:9F L:62)... still nothing...
I know my FTDI chip is genuine because I bought it from digikey and built the board myself.

I don't know what else to try.

Could you load ft_prog and show me your FT232R EEPROM settings? It's all I can think of at this point. If I unplug the FT232R while waiting for a response to the device command I get this:

Unconnected.                            > device
SerialRead expected 1 bytes, received 0 bytes from ttyUSB0
Backtrace:
  SerialRead+0xe6
  DwSerialRead+0x34
  DwReadByte+0x1d
  DwSync+0x17
  DwBreak+0x18
  TryConnectSerialPort+0x5d
  ConnectSerialPort+0x6c
  DeviceCommand+0x3b
  HandleCommand+0x7c
  ParseAndHandleCommand+0x89
  UI+0x88
  Program+0x19
Couldn't find a DebugWIRE device.
Backtrace:
  ConnectSerialPort+0x62
  DeviceCommand+0x3b
  HandleCommand+0x7c
  ParseAndHandleCommand+0x89
  UI+0x88
  Program+0x19
Unconnected.                            >

Any ideas?

dcwbrown, 2016-08-27:
If there's a specific setting you're looking for, let me know. Here's the EEPROM hex:
```
Device: 0 [Loc ID:0x1513] 

Word  MSB
0000: 4000 0403 6001 0000 2DA0 0008 0000 0A98   @...`...-.......
0008: 20A2 12C2 1023 0005 030A 0046 0054 0044    ....#.....F.T.D
0010: 0049 0320 0046 0054 0032 0033 0032 0052   .I. .F.T.2.3.2.R
0018: 0020 0055 0053 0042 0020 0055 0041 0052   . .U.S.B. .U.A.R
0020: 0054 0312 0041 0035 0030 0032 0038 0035   .T...A.5.0.2.8.5
0028: 0042 0049 D23B 50B0 0000 0000 0000 0000   .B.I.;P.........
0030: 0000 0000 0000 0000 0000 0000 0000 0000   ................
0038: 0000 0000 0000 0000 0000 0000 0000 3E8F   ..............>.
0040: 041A FBE5 0000 D23B 50B0 0042 0000 0000   .......;P..B....
0048: 0000 0000 0000 0000 4135 5754 4334 3655   ........A5WTC46U
0050: 0000 0000 0000 0000 0000 0000 0000 0000   ................
0058: 0000 0000 0000 0000 0000 0000 0000 0000   ................
0060: 0000 0000 0000 0000 0000 0000 0000 0000   ................
0068: 0000 0000 0000 0000 0000 0000 0000 0000   ................
0070: 0000 0000 0000 0000 0000 0000 0000 0000   ................
0078: 0000 0000 0000 0000 0000 0000 0000 0000   ................
```

dcwbrown, 2016-08-27:
Just wondering if you would mind trying with one of the cheap ft232 boards on Amazon? The first one I see listed on Amazon US (HiLetgo at $5.69) looks exactly like the one I have working right now. (I'm assuming you are in the US)

doubleodoug, 2016-08-27:
Hmm... Your EEPROM is different than mine... Could you save the xml template from FT_PROG? I can't hex edit the EERPOM but I can upload a template.

I have an FT 230x I'll test tonight, if neither that nor updated EEPROM settings work I'll buy one of those ft232r boards.

dcwbrown, 2016-08-28:
Here's the template attached as a file:
[ft232template.xml.txt](https://github.com/dcwbrown/dwire-debug/files/441096/ft232template.xml.txt)

It looks like this:

```
<?xml version="1.0" encoding="utf-16"?>
<FT_EEPROM>
  <Chip_Details>
    <Type>FT232R</Type>
  </Chip_Details>
  <USB_Device_Descriptor>
    <VID_PID>0</VID_PID>
    <idVendor>0403</idVendor>
    <idProduct>6001</idProduct>
    <bcdUSB>USB 2.0</bcdUSB>
  </USB_Device_Descriptor>
  <USB_Config_Descriptor>
    <bmAttributes>
      <RemoteWakeupEnabled>true</RemoteWakeupEnabled>
      <SelfPowered>false</SelfPowered>
      <BusPowered>true</BusPowered>
    </bmAttributes>
    <IOpullDown>false</IOpullDown>
    <MaxPower>90</MaxPower>
  </USB_Config_Descriptor>
  <USB_String_Descriptors>
    <Manufacturer>FTDI</Manufacturer>
    <Product_Description>FT232R USB UART</Product_Description>
    <SerialNumber_Enabled>true</SerialNumber_Enabled>
    <SerialNumber />
    <SerialNumberPrefix>A5</SerialNumberPrefix>
    <SerialNumber_AutoGenerate>true</SerialNumber_AutoGenerate>
  </USB_String_Descriptors>
  <Hardware_Specific>
    <HighIO>false</HighIO>
    <D2XX>false</D2XX>
    <ExternalOscillator>false</ExternalOscillator>
    <Invert_RS232_Signals>
      <TXD>false</TXD>
      <RXD>false</RXD>
      <RTS>false</RTS>
      <CTS>false</CTS>
      <DTR>false</DTR>
      <DSR>false</DSR>
      <DCD>false</DCD>
      <RI>false</RI>
    </Invert_RS232_Signals>
    <IO_Controls>
      <C0>TXLED#</C0>
      <C1>RXLED#</C1>
      <C2>TXDEN</C2>
      <C3>PWREN#</C3>
      <C4>SLEEP#</C4>
    </IO_Controls>
  </Hardware_Specific>
</FT_EEPROM>
```

doubleodoug, 2016-08-28:
I think I figured it out!

I wrote a python script to send 0x00 (BREAK) and read the response, then iterated over every baudrate from 0 to 1,000,000. I get valid responses (0x55) for 59200 baud to 59600 baud, when 8MHz/128 should be 62500 baud. **I'm pretty sure this is the variation in the RC oscillator.** Did you calibrate your oscillators? I do not.

This would take some programming to work around since I don't think there's a better way to detect the actual baud rate except trial and error...

I also noticed you are using 1000000/125 for your BaudRate in GlobalData.c, http://www.ruemohr.org/docs/debugwire.html and https://en.wikipedia.org/wiki/DebugWIRE say that should be a divisor of 128? Not sure if this means that documentation is wrong or what...

But this is also roughly 8,000 baud, so ~1/8th of what I've found to work? Is BaudRate in GlobalData.c actually a byte rate?

dcwbrown, 2016-08-28:
Great! No I don't calibrate my oscillators - all the chips (about half a dozen) I've tried have 'just worked'.

So have you managed to get dwdebug working by adjusting BaudRate in GlobalData.c?

Thanks for pointing out the incorrect divisor - I'll change it to 128. 

Looking at the graph on page 170 of the attiny25/45/85 doc it looks like the clock speed is indeed about 5% slower at 3v than at 5v, and if my arithmetic is right, that's the difference between 59400 (midpoint of your working baud rates) and 62500 baud.

So it probably does make sense to calibrate the oscillator.

As you say, detecting the actual rate by trial and error could be a tedious programming job. Maybe I'll get around to it :-). 

Very pleased to hear you are making progress! -- Cheers -- Dave.

closed 2016-08-28 by dcwbrown