dcwbrown/dwire-debug #3: Doesn't work for attiny13

kadamski, 2017-01-22:
While this software seems to be working reliably for attiny85, I can't really get it to work with Attiny13. I did change the Baudrate to 1200000/131 (the largest value that still works) and added an entry to characteristics:
    {0x9007,  64,   64,   64,  1024, 0x2E,  32, "ATtiny13"},

Simple commands work as the device is recognized but I can't load a binary. It always hangs on read while trying to execute this huge command from writing first page. It writes 439 bytes to serial, then reads 200, 200, 8 and hangs on trying to read remaining 31.

kadamski, 2017-01-22:
Filing a bug is just like asking someone - just forming the question can help in finding the answer. And here it is - DWDR has different address in tiny13, it's 2E, not 22, so WriteFlash.c has to be changed like this:

```
-      0xD2, 0xB4, 0x02, 0x23, *(p++),    // in r0,DWDR (low byte)
-      0xD2, 0xB4, 0x12, 0x23, *(p++),    // in r1,DWDR (high byte)
+      0xD2, 0xB4, 0x0e, 0x23, *(p++),    // in r0,DWDR (low byte)
+      0xD2, 0xB4, 0x1e, 0x23, *(p++),    // in r1,DWDR (high byte)

```

And attiny13a can also be flashed. Now I would need to form this up into a PR, I guess.

dcwbrown, 2017-01-22:
Wow, I haven't done anything with this for a couple of years, thanks for taking an interest, I hope it is of some use to you.

Great that you're getting the attiny13 to work.

Clearly that's my bug in that I'm not using the DWDR configuration value in WriteFlash code. 

It looks like when I went through the code making DWDR configurable I missed those two constructors in WriteFlash - rather than changing them to 12 I hope you can just change them to DWDR() (defined in DwPort.c included earlier by dwdebug.c - I know, a shockingly non-standard way of writing C programs.)?

Yes. It certainly looks like I never got around to implementing setting the one and only breakpoint although it seems I made a start in that the go command detects go stopping on a break. I suspect there may be more to do there, and for trace though.

All the best -- Dave.

kadamski, 2017-01-23:
I've seen that there are no recent activity in this project which is a shame because it really is cool. I don't understand how it is possible it's not popular. 

I'm not sure how to handle baud rate yet - setting proper value is clearly tricky (probably because of internal oscillator accuracy problems)  as it should be /128 of the clock but this doesn't work for both attiny85 (you are using /125 diveder) and attiny13 (I have to use /131).

But I would like to fix at least the second issue - hardcoded DWDR. One thing I don't undestand is why you are setting DWDR address to 0x42 in Characteristics even though datasheet says it's 0x22. Is this a typo?

dcwbrown, 2017-01-23:
DWDR in the configuration data is used to control which memory addresses we try to read - if you try to read the DWDR registerit causes problems (i think it hung) because it messes with the dwire process itself.

DWDR is at offset 0x22 in the io registers. but that maps as offset 0x42 in data memory, because in data memory the io registers are preceeded by plain old r0 - r31. 

So the 0x42 wascorrect for the purpose I was using it for. 

I'm updating the flash memory code to take this into account. All being well.

I'm afraid that 125 divider may have been another typo. I don't quite remember. But really the code should be trying multiple baud rates because running the chips at 3.3 instead of 5 volts for example changes the frequency a bit too much. Probably not too difficult ....

Thanks for calling it cool :-).

dcwbrown, 2017-01-23:
OK, I've just committed:
 -- Added your attiny config line to GlobalData (many thanks)
 -- Fixed WriteFlash to use the right DWDR register for each chip
 -- Fixed the baud rate divisor (was 125, should have been 128)

I changed the dwdr def in the config from data address to io reg number as it was confusing before, and added separate DWDRaddr() and DWDRreg() functions to get the right values.

Naughty me, I haven't actually tested the updated WriteFlash code yet - I need to move to another machine. Please let me know if it works for you!

Assuming this works I'll look at adding the breakpoint command.

Thanks -- Dave.

kadamski, 2017-01-23:
Flashing works good on both Attiny13 and Attiny85 after some baudrate tweaking (128 divder doesn't work good for my attiny85, again it's probably because of calibration thing).

dcwbrown, 2017-01-23:
That's good news. Thanks.

I've just committed breakpoint support. It turned out I had done most of the hard work before.

There is just one breakpoint, but unlike the Atmel tools, it is not implemented by reflashing the chip, so it's fast and doesn't wear out the flash.

Would love to hear if it works well for you.

Re divider - perhaps I should put it back to 125. What voltage was your ATtiny85 running at?

All the best -- Dave.

dcwbrown, 2017-01-23:
Some todo's for me ...

 - Update manual.md to reflect all the things that have been implemented.
 - Document support for line numbers and symbols in ELF files and the bellpush sample provided to show how to use this.
 - Link to manual from readme
 - Implement multi-instruction trace
 - Add eeprom dumping and editing commands
 - Show presence of active breakpoint in UI somehow
 - Provide way to enable/disable timers

The last point is interesting - it appears that the go command can start the device with timers either enabled or disabled. Currently I have the flag set to disable them. I don't actually know if it works.

I should probably leave them enabled by default, and provide a command to let the user run with them disabled if they want that.

plasticassius, 2017-01-23:
I hope it's ok if I jump into the conversation here. I just ran across this project today, and I agree that it is cool. I haven't gotten it working yet, but after reading this thread, I imagine that it must be a baud rate issue. I'm using an Attiny84 and my code uses Vusb, so my clock speed is nonstandard. I'll look into that as well as getting a proper ft232r which I ordered today, but will take some time to get delivered.

A question I have as a result of building the project is as a cli program does it actually use gtk+-3.0 and pkg-config? If not, can the dependency be removed? It's an enormous install. (I don't have any experience with these packages so please excuse the question if it's silly).

kadamski, 2017-01-23:
@dcwbrown: The breakpoint functionality seems to be working great for both attiny13 and attiny85 and makes the project even cooler. I'm not sure, however, if its that intuitive to give breakpoint address in words, not in actual address in memory.  As for the getting back to 125 baudrate divider.. I don't think this is a good idea. I guess there is just no value that will work for everybody so you either should support setting custom baudrate from command line or think about autodetection (or both). Autodetection can be tricky, however, as my experience shows that there are multiple baudrates which seems to work most of the time (like they will return 55 after reset) but may still fail more often than others from time to time. Like, for example, for my Attiny13A, 1200000/131 works most of the time but fails from time to time (it will then print something like "WriteDebug, byte 10 of 10: Read 00 expected 20") but 1200000/132 seems to be more reliable.

kadamski, 2017-01-23:
@dcwbrown: and sorry to tell you that but your commits looks like rubish here in github because of the endline issues you have in your editor. Fortunately I can look at them when using "-w" option to "git log" command :)

dcwbrown, 2017-01-24:
@kadamski: Frustrating - that's the difference between LF and CRLF for eoln. I still find this a pain as I mostly work on cygwin under windows. 

Did you know that as well as using -w locally, you can add '?w=1' at the end of the github commit url for the same effect? E.g.:

```
https://github.com/dcwbrown/dwire-debug/commit/611710387cb7fd790c321484de8621b850733942?w=1
```

dcwbrown, 2017-01-24:
@kadamski: I thought a lot about code addresses - whether they should be word or byte addresses and settled on word addresses because that's what symbol values and jump offsets are calculated in.

However since you raise the question I've looked again, and I see avr-as listing files use byte addresses to label lines.

So I'll add this to a list of things to change - it's not just the B command, also the PC command and the addresses shown by unassemble.

Hmm, for the PC command I guess it needs to remain a word address, because that is the value stored in the PC register.

Then the question is should the PC command and B command be different?

dcwbrown, 2017-01-24:
@kadamski: I guess I need to add baud rate hunting to the todo list.

dcwbrown, 2017-01-24:
@plasticassius: GTK is only used for its file chooser dialog, which is shown when you use the L command without a file name. It's really nice to have, but it can easily be omitted. I'll look at adding a NOFILECHOOSER define in dwdebug.c.

kadamski, 2017-01-27:
I guess this one can be closed now as attiny13 is working already :)

closed 2017-01-27 by kadamski