dcwbrown/dwire-debug #4: Add baudrate hunting

dcwbrown, 2017-01-24:
When an ft232 is found we should take some time to find the right baud rate.

It may be possible to put the ft232 into bit-bang mode, send a break, and then measure the time periods of the pulses in the 0x55 that should be returned.

dcwbrown, 2017-01-24:
Hmm, bitbang mode needs a special driver on Linux. Let's try without bitbang mode first.

plasticassius, 2017-01-24:
I don't know about programming an ft232, but I think this should work well on an Attiny85. You could use the Attiny85 in place of the ft232 for about the same price and without special drivers for Linux or Windows. Let me think about this and get back to you. Would you point me to where I can find in detail what you mean by "send a break, and then measure the time periods of the pulses in the 0x55?"

I haven't worked with serial port data in great detail, but I'm guessing that a break is an rs232 thing. Are the other parameters one stop bit, no parity? I imagine I can find a description of rs232 without a big effort, but if you have handy anything about it specifically related to DebugWire, I'm interested.

dcwbrown, 2017-01-24:
A break is when the line is held at zero for a much longer period than the time it takes to transmit a byte. 

ATtiny dwire handling recognises this and stops the ATtiny processor when it sees it, and responds by sending a 0x55 byte back up the dwire connection. 

The dwire protocol follows rs232 timing with one start bit, no parity, and one stop bit.

It turns out that the special FT232 modes aren't necessary, the following works fine with the stadard OS built-in serial-over-USB drivers:

Having sent a break by holding the line at zero, reading back from the FT232 one receives first a load of 0 bytes corresponding to the low line, then possibly an 0xFF byte corresponding to the line going high, then the 0x55. 

0x55 is the nice bit pattern  0 1 0 1 0 1 0 1.

If the timing is wrong, then instead of 01010101, we see runs of more than one adjacent 0 or 1 (e.g. if the baud rate is set too fast), or multiple zeroes at the end (e.g. if the baud rate is set too slow). 

Sadlythese cases are not exclusive, and I'm currently experimenting to come up with a good algorithm to determine the next best baud rate guess based on how the received 10101010 has been mangled.

---

The great advantage of the FT232 is that you can buy it easily and at a reasonable price, and the cable wit the diode is very easy to make. 

You certainly could use an attiny 45 instead, running the bit-banged tiny-USB code, but then you need a board with usb and power handling designed, and you are somewhat at the mercy of the software USB implementation.

plasticassius, 2017-01-24:
[Digispark USB Development Board](http://digistump.com/products/1) are priced comparably to FT232, and I've had good results running the software USB code on them. I've also used the uart on the attiny85 with these. Also, it shouldn't be a problem to measure the timing of the 0x55 with a timer on the attiny85. I think this is an option to keep in mind if the FT232 has difficulties; however, never mind if the FT232 works for you. I'd be willing to try this with the attiny85 if it turns out that you'd be interested.

dcwbrown, 2017-01-24:
Mmmm. Nice simple board. And it looks like clones are available wonderfully cheaply on ebay:
```
http://www.ebay.co.uk/itm/10Pcs-Digispark-Kickstarter-Attiny85-Micro-USB-Development-Board-for-Arduino-/302165129644
```
10 for £15.79. (I'm in the UK)

Yes, if e.g. you use a pin change interrupt you can get accurate time measurements of the pulses coming back after the break and determine the baud rate accurately.

I'll spend a day or so with the FT232 code and see if I succeed.

plasticassius, 2017-01-25:
I looked into it some more, and found that while the attiny85 has support for various serial protocols like i2c, it doesn't implement a full uart. However, with timer support, it looks straightforward to implement all of the particulars of debug wire including putting all of the functionality on a single pin. Thus, it should be possible to do away with the need for the diode and make an elegant solution.

That said, this seems to be a bit more work than searching for the correct baud rate with an FT232. But, I'm still willing to work on it as time allows. In addition to the datasheet, here are some related links I've found: [http://www.technoblogy.com/show?RPY](http://www.technoblogy.com/show?RPY) and [https://tinusaur.org/2015/10/14/new-library-usitwix-using-usi-as-twi-i2c/](https://tinusaur.org/2015/10/14/new-library-usitwix-using-usi-as-twi-i2c/)

dcwbrown, 2017-01-25:
I have just committed what I hope/believe is a good enough baud rate search.

I've left trace messages in for now to help debugging if it doesn't work. It may need to be less agressive in coming down to the right approx baud rate in the first step.

Also I haven't tested in Linux.

It does of course take a short time. It's not bad, and would probably feel faster without the trace messages. 

Arguably it could write out the final discovered baud rate to a config file and use that next time. I'm hoping to avoid that complexity though.

Please would you let me know whether it works for your chips?

plasticassius, 2017-01-25:
You're certainly fast, actually too fast for me. I don't have my hardware yet.

It doesn't work; however, I don't have an FT232. I ordered one two days ago when I first came across your code, so it should get to me any day now. The closest thing I have right now is a 067b:2303 Prolific Technology, Inc. PL2303 Serial Port. Here is what I got:

`dwdebug

Unconnected.                            > device /dev/ttyUSB0
Trying ttyUSB0, baud rate 50000, No response, giving up.
Couldn't find a DebugWIRE device.
Backtrace:
  dwdebug(ConnectSerialPort+0x62) [0x40809c]
  dwdebug(DeviceCommand+0x3b) [0x4080ee]
  dwdebug(HandleCommand+0x7c) [0x40ace9]
  dwdebug(ParseAndHandleCommand+0x98) [0x40adb1]
  dwdebug(UI+0x88) [0x40afa1]
  dwdebug(Program+0x19) [0x40afe4]
  dwdebug(main+0x68) [0x404210]
  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7fec5d029830]
  dwdebug(_start+0x29) [0x403c19]
*** stack smashing detected ***: dwdebug terminated
Aborted (core dumped)
`
I guess it doesn't make sense to read too much into this since it's not the right kind of device. I'll let you know what I come up with once I have the correct hardware.

kadamski, 2017-01-25:
It works great with my attiny13a but hangs on read() syscall when using attiny85:
```
$ ./dwdebug
Unconnected.                            > r
Trying ttyUSB0, baud rate 50000, received 11100000, factor 430%
Trying ttyUSB0, baud rate 11764, received 11100000, factor 430%
```

here's how it looks on strace:
```
write(1, "Trying ttyUSB0, baud rate 11764,"..., 64Trying ttyUSB0, baud rate 11764, received 11011011, factor 160%
) = 64
open("/dev/ttyUSB0", O_RDWR)            = 4
ioctl(4, TCSETS2, {0x1000 /* B??? */ -opost -isig -icanon -echo ...}) = 0
ioctl(4, TCSBRKP, 1)                    = 0
read(4, "\375", 1)                      = 1
close(4)                                = 0
write(1, "Trying ttyUSB0, baud rate 07589,"..., 64Trying ttyUSB0, baud rate 07589, received 11111101, factor 530%
) = 64
open("/dev/ttyUSB0", O_RDWR)            = 4
ioctl(4, TCSETS2, {0x1000 /* B??? */ -opost -isig -icanon -echo ...}) = 0
ioctl(4, TCSBRKP, 1)                    = 0
read(4, "\0", 1)                        = 1
read(4,
```

So it never actually gets that 0x55.. or any other response.

dcwbrown, 2017-01-25:
Ah, you are both using linux. I'll debug on linux tomorrow.

@plasticassius, that's interesting that it worked at all on the PL2303. Maybe it would work alright if it starts at a lower baud rate - DeviceCommand.c linw 177 is the initial baud rate, you could try lowering that value to see if there is a PL2303 max below which it will work?

@kadamski, Hey, that's great that it works on your attiny13. My attiny85 is working on windows. I'll try and repro on linux tomorrow.

plasticassius, 2017-01-25:
I think I have other problems here. When I run device, it usually hangs with no output at all. I've tried to put setbuf(stdout, nullptr); setbuf(stderr, nullptr); at the start of main to make output unbuffered, but I can't get any output from Ws() at all. The output I pasted above showed up only once, and after a long delay (10 min or so). Something is clearly not communicating properly. I'll get back to this later.

dcwbrown, 2017-01-26:
FYI looks like the linux serial over usb code needs more work, currently attempting to ensure the port is flushed properly before breaking. Foe xample,  one googles earch said do a sleep(2) before tcflush. This actually wroks, but makes things dreadfully slow. 

Progress is happening. Hold on ...

plasticassius, 2017-01-26:
I got my FT232 today and the output is:
```
dwdebug
Unconnected.                            > device
Trying ttyUSB0, baud rate 50000, received 10101101, factor 120%
Trying ttyUSB0, baud rate 43478, received 11011011, factor 160%
Trying ttyUSB0, baud rate 28050, received 11110000, factor 330%
```
after that, it hangs. I'm connecting to an Attiny84@12.8MHz so the baud rate should be 10KHz if I understand correctly.

dcwbrown, 2017-01-26:
I have committed a change that relies on a 10ms usleep. Not nice at all. Apparently the ioctl(TCFLSH,...) cannot control buffering within the USB device. By calling usleep we allow the USB device to finish messing about so the TCFLSH can do its job fully. Difficult to feel confident this will always work. 

Nonetheless, could you give it a try and let me know if it works for you?

Thanks!

plasticassius, 2017-01-26:
In this version I get
```
dwdebug
Unconnected.                            > device
Trying ttyUSB0, baud rate 20000, breaklen 0050, skipping [0], received 11111111, factor 730%
Trying ttyUSB0, baud rate 02777, breaklen 0036Couldn't open serial port.
Backtrace:
```
As an additional issue, I don't see any debug messages (ie: Trying...) until I pull the serial device out of the USB connector and thereby cause the program to exit with a stack trace.

dcwbrown, 2017-01-26:
That's disappointing :-(.
I think the 11111111 means that the attiny didn't respond. Hmm, it's just possible your device has become disconnected? But more likely the break is not being generated correctly. I'm using a break length parameter that seems to work for me but others have had trouble with. I'll try setting it to 0 (for default behaviour which is said to be more reliable). Will commit again soon.

plasticassius, 2017-01-26:
As I look at SimpleOutput.c I'm becoming convinced that the buffering is hiding messages. Also, I think that the system level buffering is already adequate if not problematic on it's own (please see edit to my previous comment).

dcwbrown, 2017-01-26:
OK, committed.

plasticassius, 2017-01-26:
```
dwdebug
Unconnected.                            > device
Trying ttyUSB0, baud rate 40000, breaklen 0050, skipping [0], received 11110010, factor 330%
Trying ttyUSB0, baud rate 12500, breaklen 0008, skipping [0], No response, giving up.
Couldn't find a DebugWIRE device.
```
and hangs until physical disconnect.

dcwbrown, 2017-01-26:
Re buffering - it-s cross platfrom code and it flushes at eol like most runtime systems. You can stick a flush in at the end of Ws or Wd to flush more often if you like - it's a tradeoff between an expensive OS call and immediate output.
The hang you were getting before was because the code was waiting for a serial byte. The last checkin worked around that.
I think I see an issue with the spoeed guessing code that is showing up on your chip but none of mine. I should not be counting the leading 1111 as a pulse width because it may be from the previous stop bit.
Thanks for keeping trying. 
If it is any comfort I can confirm it's working OK here. So we should not be without hope ...
Working on it.

dcwbrown, 2017-01-26:
OK, I have another commit. Sill works for me. For your scenario I believe I have corrected an overenthusiastic baud rate reduction coming from misreading 1's coming from the previous stop bits as a valid indication of pulse width. It though it was seeing a pulse wider than was really there.
Thanks ....

dcwbrown, 2017-01-26:
(Just committed again for a corner case in the last fix. Crikey.)

plasticassius, 2017-01-26:
```
Unconnected.                            > device
Trying ttyUSB0, baud rate 40000, breaklen 0050, skipping [0], received 11110010, factor 200%
Trying ttyUSB0, baud rate 21052, breaklen 0004, skipping [0], No response, giving up.
Couldn't find a DebugWIRE device.
```

dcwbrown, 2017-01-26:
Oh dear. No difference. 
The break is echoing back as a 0, but the chip isn't generating a 0x55. Which suggests the chip isn't seeing it as a break. Which suggests it is not long enough. 

On windows the api doesn't have a break instruction, instead it allows you to force the line low, then you usleep however long you want, then release the line. Which gives better control of the break length. 

Interestingly though the first break does work on your machine.

So inserting a delay before doing subsequent breaks might work, but gosh, I'm really mucking about in the dark, using you as a guinea pig. 

How are feeling about keep trying what are effectively hacks?

Which version of linux are you using?

plasticassius, 2017-01-26:
To me it looks like the second read gets no response at all and hangs until I put the device out of the USB connector although the first read looks like it at least returns some data. I checked the connections, and that looks ok. I also tried a loopback test with cat and rx connected to tx, that seems ok. Could there be something else in my setup?

I'm using ubuntu 16.04. I don't mind trying random stuff, but perhaps I can think of something to try if the thing that's missing is being able to generate a break.

dcwbrown, 2017-01-26:
The [0] means that we received a zero back on the first ReadByte. It's the second one that gets nothing.
I've committed another hack to pause before breaking in the hope it lets the chip recognise the break.

Sorry - I have to go help a neighbour whose lights have just gone out - should be back in 15 mins.

plasticassius, 2017-01-26:
```
Unconnected.                            > device
Trying ttyUSB0, baud rate 40000, breaklen 0050, skipping [0], received 11110010, factor 200%
Trying ttyUSB0, baud rate 21052, breaklen 0050, skipping [0], No response, giving up.
```
To be specific: I have an FT232RL and I've set the jumper to 3.3V and I'm using it to power the Attiny84. I'll go ahead and try an Attiny85 in case that matters.

dcwbrown, 2017-01-26:
I'm on Ubuntu 16.10 64 bit. It seems unlikely there would be a difference. 
Next thing to try would perhaps be a whole second of sleep between attempts. 
First, yes, I'd like to know what symptoms you have with attiny85.

plasticassius, 2017-01-26:
I have Ubuntu 64 bit also. In the process of setting up my t85, I realized that while it's clear I need to enable the debug wire fuse, do I need to also disable the external reset fuse? On my t84 I have debug wire enabled, but I left external reset enabled. Could this be the problem? Now that I'm thinking about it, the "break" could have been just resetting the t84. In other words, should I set hfuse to 0x95 or 0x15? Sorry that I lead you to run through all of these hacks if that was in fact the problem.

kadamski, 2017-01-26:
Did you try using different ioctl? Like TIOCSBRK then sleep then TIOCCBRK? Or using nonzero argument to TCSBRKP?

dcwbrown, 2017-01-26:
@plasticassius: I just found a fresh attiny84 and enabled dwire: hfuse = 9f. And it's working fine from my Ubuntu.
So yes, you should leave external reset enabled. The bottom bits are interesting though - you have a enabled brownout level, and you have eeprom preserve through erase set. 
It could be the brownout level - the ft232 provided voltage may well be going outside the range 2.5 .. 2.9 that BOD 5 sets. Could you try again with 9f?

dcwbrown, 2017-01-26:
@kadamski: Thanks very much for those. I have found it very difficult indeed to get up to date docs on serial programming, especially with struct termios2. I hadn't found TIOCSBRK/TIOCCBRK. If I can use those the code will be much more the same between Linux and Windows which should hopefully mean fewer discrepancies. Will look at that.

kadamski, 2017-01-26:
That's interesting.. using TCSBRKP with argument of 0, as you are using now, should genetate a break for 0.25s. And a resolution of the argument is very low, you set it in deciseconds so the shortest break you can get is 0.1s when you use 1 as argument. I think even that is  a lot.

kadamski, 2017-01-26:
This: http://lxr.free-electrons.com/source/drivers/tty/tty_io.c#L2954 is the most up to date documentation. But http://man7.org/linux/man-pages/man4/tty_ioctl.4.html isn't bad either :)

kadamski, 2017-01-26:
You can also use https://linux.die.net/man/3/tcsendbreak which is more portable (POSIX specified). On linux it's essentialy the same as using TCSBRKP ioctl, though.

dcwbrown, 2017-01-26:
Just committed change to use TIOCSBRK/TIOCCBRK. Yes, it makes baud rate detection much faster. It works on the 3 chips I have here now - attiny85, 13 and 84 - all running using their as-supplied internal oscillator.
Thanks for the document references.

plasticassius, 2017-01-26:
Here's what I get with my t85 with hfuse = 9f:
```
Trying ttyUSB0, baud rate 40000, breaklen 0050, skipping [0], No response, giving up.
Couldn't find a DebugWIRE device.
```
and with my t84 with hfuse = 95:
```
Unconnected.                            > device
Trying ttyUSB0, baud rate 40000, breaklen 0050, skipping [0], received 11110010, factor 200%
Trying ttyUSB0, baud rate 21052, breaklen 0050, skipping [0], No response, giving up.
Couldn't find a DebugWIRE device.
```
I don't know what else to try at this point. I'm going to take a break for some dinner.

kadamski, 2017-01-26:
@plasticassius: Your final bandwidth should be around clock/128. Fiddle around with starting baud rate close to what you would expect and see if that works for you. It's in DeviceCommand.c:FindBaudRate()

@dcwbrown: Current code works like a charm with t85 but not with t13. Here's what I get:
```
./dwdebug f0,q
Trying ttyUSB1, baud rate 40000, breaklen 0050, skipping [0], received 11110000, factor 230%
Trying ttyUSB1, baud rate 18181, breaklen 0050, skipping [0], received 01100110, factor 200%
Trying ttyUSB1, baud rate 09568, breaklen 0050, skipping [0], received 10010101, factor 125%
Trying ttyUSB1, baud rate 08320, breaklen 0050, skipping [0], received 11010101, factor 110%
Trying ttyUSB1, baud rate 07923, breaklen 0050, skipping [0], received 10101101, factor 125%
Trying ttyUSB1, baud rate 06889, breaklen 0050, skipping [0], received 11011011, factor 133%
Trying ttyUSB1, baud rate 05600, breaklen 0050, skipping [0], received 11101110, factor 230%
```

If I change starting baudrate to 9000, however, I get:
```
Trying ttyUSB1, baud rate 09000, breaklen 0050, skipping [0], received 01010101: expected result.
Finding upper bound.
Trying ttyUSB1, baud rate 09180, breaklen 0011, skipping [0], received 01010101: expected result.
Trying ttyUSB1, baud rate 09363, breaklen 0011, skipping [0], received 01010101: expected result.
Trying ttyUSB1, baud rate 09550, breaklen 0011, skipping [0], received 10010101
Finding lower bound.
Trying ttyUSB1, baud rate 08820, breaklen 0011, skipping [0], received 01010101: expected result.
Trying ttyUSB1, baud rate 08643, breaklen 0011, skipping [0], received 01010101: expected result.
Trying ttyUSB1, baud rate 08470, breaklen 0011, skipping [0], received 01010101: expected result.
Trying ttyUSB1, baud rate 08300, breaklen 0011, skipping [0], received 11010101

Chosen baud rate 8916
```

dcwbrown, 2017-01-26:
OK, thanks. Looks like I need to reduce the baud rate less aggressively.  The third step generated 10010101 which is clearly pretty close - yet I chose to try approx a 15% reduction when a 5% would have been more appropriate. 
Time to clean up the calculations.
Re clock/128 - I think it's more like /1024: that 8916 baud corresponds to a default internal clock, which I believe is 8MHz?

kadamski, 2017-01-26:
Nope, the default clock of t13 is 1.2Mhz and 1Mhz fir t85.

plasticassius, 2017-01-26:
@dcwbrown Thanks for bearing with me on this.

Right now I can't imagine anything that would have made me feel dumber than this lesson in arithmetic: 12.8MHz / 128 = 100KHz. So, it turns out I needed to start higher than 50KHz. Here's what I got on my t84:
```
Trying ttyUSB0, baud rate 150000, breaklen 0050, skipping [0], received 10001100, factor 230%
Trying ttyUSB0, baud rate 68181, breaklen 0050, skipping [0], received 10101001, factor 120%
Trying ttyUSB0, baud rate 61982, breaklen 0050, skipping [0], received 10100101, factor 125%
Trying ttyUSB0, baud rate 53897, breaklen 0050, skipping [0], received 01010101: expected result.
Finding upper bound.
Trying ttyUSB0, baud rate 54974, breaklen 0001, skipping [0], received 01010101: expected result.
Trying ttyUSB0, baud rate 56073, breaklen 0001, skipping [0], received 01010101: expected result.
Trying ttyUSB0, baud rate 57194, breaklen 0001, skipping [0], received 01010101: expected result.
Trying ttyUSB0, baud rate 58337, breaklen 0001, skipping [0], received 01010101: expected result.
Trying ttyUSB0, baud rate 59503, breaklen 0001, skipping [0], received 01010101: expected result.
Trying ttyUSB0, baud rate 60693, breaklen 0001, skipping [0], received 10010101
Finding lower bound.
Trying ttyUSB0, baud rate 52819, breaklen 0001, skipping [0], received 10010101

Chosen baud rate 56700

, skipping [0]ATtiny84 found on ttyUSB0
```
Since it found 56.7KHz (*128=7.26MHz) I must have forgotten what was loaded on this part since that is close to 8MHz as set by the fuses.
On my t85:
```
Trying ttyUSB0, baud rate 150000, breaklen 0050, skipping [0], received 10010010, factor 166%
Trying ttyUSB0, baud rate 96153, breaklen 0050, skipping [0], received 01010101: expected result.
Finding upper bound.
Trying ttyUSB0, baud rate 98076, breaklen 0001, skipping [0], received 01010101: expected result.
Trying ttyUSB0, baud rate 100037, breaklen 0001, skipping [0], received 01010101: expected result.
Trying ttyUSB0, baud rate 102037, breaklen 0001, skipping [0], received 01010101: expected result.
Trying ttyUSB0, baud rate 104077, breaklen 0001, skipping [0], received 01010101: expected result.
Trying ttyUSB0, baud rate 106158, breaklen 0001, skipping [0], received 11010101
Finding lower bound.
Trying ttyUSB0, baud rate 94229, breaklen 0001, skipping [0], received 01010101: expected result.
Trying ttyUSB0, baud rate 92344, breaklen 0001, skipping [0], received 10010101

Chosen baud rate 99153
```
This is very close to the 100KHz I expected since I calibrated this part to 12.8MHz. However, I usually run t85s at 16.5MHz since the bit banged USB works best at that speed. That would correspond to 129KHz. I think it's important to keep in mind that speeds upwards of 20MHz are possible, so baud rates could range higher than 156KHz.

To me it seems that the calibration runs quite quickly.

dcwbrown, 2017-01-26:
I am probably confusing the clock and the oscillator? It's the internal oscillator that's 8MHz?

I've committed a simpler baud rate hunting algorithm which should approach the target speed more carefully. Hopefully it will be better.

plasticassius, 2017-01-27:
It didn't work for the t85:
```
Trying ttyUSB0, baud rate 150000, breaklen 0050, received 10010010, scale 95%
Trying ttyUSB0, baud rate 142500, breaklen 0050, received 11011010, scale 60%
Trying ttyUSB0, baud rate 85500, breaklen 0050, received 10101001, scale 95%
Trying ttyUSB0, baud rate 81225, breaklen 0050, received 11101001, scale 95%
Trying ttyUSB0, baud rate 77163, breaklen 0050, received 11001011, scale 60%
Trying ttyUSB0, baud rate 46297, breaklen 0050, No response, giving up.
, scale 0%
Trying ttyUSB0, baud rate 00000, breaklen 0050Couldn't open serial port.
```
The previous commit worked well, and quickly too.

According to the t85 data sheet, the internal oscillator is 8 MHz. But up to 20 MHz clock is considered "safe operating." I think that up to 24 MHz typically works.

dcwbrown, 2017-01-27:
Some more commits, fixing that case I believe.
Off to bed now, I think it's pretty good.

kadamski, 2017-01-27:
The internal oscilator is 8Mhz on t85 and 9.6Mhz on t13 but by default there is a div8 fuse programmed making tge actual clock 1Mhz/1.2Mhz.

plasticassius, 2017-01-27:
@dcwbrown  Looks good!
```
Trying ttyUSB0, baud rate 150000, breaklen 0050, received 10010010 7 032000000, scale 85%
Trying ttyUSB0, baud rate 127500, breaklen 0050, received 01001011 8 042000000, scale 85%
Trying ttyUSB0, baud rate 108375, breaklen 0050, received 10100101 7 051000000, scale 95%
Trying ttyUSB0, baud rate 102956, breaklen 0050, received 01010101: expected result.
Finding upper bound.
Trying ttyUSB0, baud rate 105015, breaklen 0000, received 01010101: expected result.
Trying ttyUSB0, baud rate 107115, breaklen 0000, received 10100101 7 051000000
Finding lower bound.
Trying ttyUSB0, baud rate 100896, breaklen 0000, received 01010101: expected result.
Trying ttyUSB0, baud rate 98878, breaklen 0000, received 01010101: expected result.
Trying ttyUSB0, baud rate 96900, breaklen 0000, received 01010101: expected result.
Trying ttyUSB0, baud rate 94962, breaklen 0000, received 01010101: expected result.
Trying ttyUSB0, baud rate 93062, breaklen 0000, received 01010101: expected result.
Trying ttyUSB0, baud rate 91200, breaklen 0000, received 10010101 7 051000000

Chosen baud rate 99038

ATtiny85 found on ttyUSB0

Trying ttyUSB0, baud rate 150000, breaklen 0050, received 10001100 7 002100000, scale 85%
Trying ttyUSB0, baud rate 127500, breaklen 0050, received 11001110 6 011100000, scale 95%
Trying ttyUSB0, baud rate 121125, breaklen 0050, received 11100110 5 012000000, scale 85%
Trying ttyUSB0, baud rate 102956, breaklen 0050, received 00110110 8 023000000, scale 85%
Trying ttyUSB0, baud rate 87512, breaklen 0050, received 10010010 7 032000000, scale 85%
Trying ttyUSB0, baud rate 74385, breaklen 0050, received 01001011 8 042000000, scale 85%
Trying ttyUSB0, baud rate 63227, breaklen 0050, received 10100101 7 051000000, scale 95%
Trying ttyUSB0, baud rate 60065, breaklen 0050, received 10010101 7 051000000, scale 95%
Trying ttyUSB0, baud rate 57061, breaklen 0050, received 01010101: expected result.
Finding upper bound.
Trying ttyUSB0, baud rate 58202, breaklen 0001, received 01010101: expected result.
Trying ttyUSB0, baud rate 59366, breaklen 0001, received 01010101: expected result.
Trying ttyUSB0, baud rate 60553, breaklen 0001, received 10010101 7 051000000
Finding lower bound.
Trying ttyUSB0, baud rate 55919, breaklen 0001, received 01010101: expected result.
Trying ttyUSB0, baud rate 54800, breaklen 0001, received 01010101: expected result.
Trying ttyUSB0, baud rate 53704, breaklen 0001, received 01010101: expected result.
Trying ttyUSB0, baud rate 52629, breaklen 0001, received 10010101 7 051000000

Chosen baud rate 56535

ATtiny84 found on ttyUSB0
```
@kadamski I didn't check, but what you say about the defaults for new parts sounds correct. However, there are many ways to select other speeds. To summarize, it is possible to use internal or external sources, select divisors (and a 2x multiplier in a t85) with fuses or by writing registers in the firmware, and to tune the clock with a calibration register. The upshot of this is that the clock speed may anything up to 20 MHz which is considered ok by the datasheet, or 24 MHz if someone pushes it. Clock speeds up to 20 MHz are both allowed and in common use.

dcwbrown, 2017-01-28:
Thank you both for all the testing and advice. 

I think this is working well enough for now.

I have committed:
 - Changed the baud hunting messages to show only a row of dots by default. Each dot is a new rate being tested. The old fulll trace messages are still available with the verbose command. (example below.)
 - The device command now accepts a baud rate, so you can bypass all the baud rate hunting if you know the correct rate. E.g.

```
./dwdebug.exe device ttyUSB0 99000
```
for @plasticassius's attiny95.

For verbose hunting messages use

```
./dwdebug.exe verbose
```

-- Dave.

dcwbrown, 2017-02-01:
Closing this as working.

Thanks again to @plasticassius and @kadamski for extensive testing and advice.

closed 2017-02-01 by dcwbrown