dcwbrown/dwire-debug #5: Make file chooser optional

dcwbrown, 2017-01-24:
(Thanks @plasticassius). 

Provide a #define to disable file chooser support on linux so that dwire-debug can be built without GTK build support installed.

plasticassius, 2017-01-24:
I certainly don't see this as high priority, and I didn't mean to complain before I got anywhere. It's just the first thing I ran into while building this code.

dcwbrown, 2017-01-25:
Have added a NOFILEDIALOG option to the Makefile - it removes the need for GTK support, but you still need pkg-config. This should be quick enough to install, e.g.:
```
sudo apt install pkg-config
```

plasticassius, 2017-01-25:
Thanks for this, it'll be nice to not have to install gtk+-3.0 on my systems. Since you've already put in the effort to support the NOFILEDIALOG option, what do you say to taking the extra step to make it really slick and autodetect gtk+-3.0? The following works for all combinations of NOFILEDIALOG defined/not defined and gtk+-3.0 loaded/not loaded:

`
--- a/Makefile
+++ b/Makefile
@@ -9,6 +9,15 @@ ifdef WINDIR
 BINARY := $(TARGET).exe
 else
 BINARY := $(TARGET)
+
+#autodetect FILEDIALOG options for Linux based on existance of gtk+-3.0
+ifndef NOFILEDIALOG
+       FILEDIALOG=$(shell pkg-config --exists gtk+-3.0 && echo \`pkg-config --cflags --libs gtk+-3.0\`)
+endif
+ifeq ($(strip $(FILEDIALOG)),)
+       FILEDIALOG=-DNOFILEDIALOG
+endif
+
 endif
 
 .PHONY: all
@@ -31,11 +40,7 @@ else
        #i686-w64-mingw32-gcc -g -oo -std=gnu99 -Wall -o $(BINARY) -Dwindows $(TARGET).c -lKernel32 -lComdlg32
 endif
 else
-ifdef NOFILEDIALOG
-       gcc -std=gnu99 -g -fno-pie -rdynamic -Wall -o $(BINARY) -DNOFILEDIALOG $(TARGET).c `pkg-config --cflags`
-else
-       gcc -std=gnu99 -g -fno-pie -rdynamic -Wall -o $(BINARY) $(TARGET).c `pkg-config --cflags --libs gtk+-3.0`
-endif
+       gcc -std=gnu99 -g -fno-pie -rdynamic -Wall -o $(BINARY) $(TARGET).c $(FILEDIALOG)
 endif
        ls -lap $(BINARY)
 `

I didn't try to change the windows case, but my guess is that dialog box support is typically installed on windows and won't be an issue there.

dcwbrown, 2017-01-28:
Closing this now since adding @plasticassious' autodetect. Thanks @plasticassious.

closed 2017-01-28 by dcwbrown