dcwbrown/dwire-debug #9: Breakpoints and timers

dcwbrown, 2017-01-24:
Show status of breakpoint setting in the UI somehow.

Provide option to disable or enable timers during execution and show that in the UI too.

plasticassius, 2017-10-25:
I see you have te and td commands (with td by default) which set the go context to enable or disable timers. But, what timers does this control? After I use td, timer0 and wdt still operate.

Also, reset still fails (also notice the changing baud rate):
```
Unconnected.                            > td
Connected to ATtiny84 on UsbTiny1 at 100609 baud.
0d2c: 91cf  pop   r28                   > td
0d2c: 91cf  pop   r28                   > g 
Running. Press return key to force break. Waiting ...a
Keyboard requested break. 
0c8a: 9180  lds   r24, $006e            > qr
$dwdebug
Unconnected.                            > ls
ATtiny84 on UsbTiny1 at 100609 baud.    
Unconnected.                            > reset
Connected to ATtiny84 on UsbTiny1 at 100609 baud.
Could not read back timings following transfer and sync command
Segmentation fault (core dumped)
$dwdebug
Unconnected.                            > r
Connected to ATtiny84 on UsbTiny1 at 62977 baud.
r0 a1   r4 7f    r8 71   r12 ed   r16 fe   r20 01   r24 62   r28 5f   
r1 00   r5 6e    r9 5f   r13 cb   r17 00   r21 a0   r25 00   r29 02   
r2 02   r6 6e   r10 f7   r14 ff   r18 18   r22 00   r26 9c   r30 b9   
r3 b9   r7 62   r11 e4   r15 bf   r19 00   r23 02   r27 00   r31 00   
SREG i t h s v n z c    PC 0000  SP 025f   X 009c   Y 025f   Z 00b9   
0000: c06a  rjmp  00d6 (+107)           > qr
$dwdebug
Unconnected.                            > device u
Connected to ATtiny84 on UsbTiny1 at 100000 baud.
05b6: 9180  lds   r24, $0095            > reset
Could not read back timings following transfer and sync command
Segmentation fault (core dumped)
```

plasticassius, 2017-11-29:
> I see you have te and td commands (with td by default) which set the go context to enable or disable timers. But, what timers does this control? After I use td, timer0 and wdt still operate.

I was puzzling over this for a while, but I see you answered my te and td question here: #31

dcwbrown, 2017-11-29:
I don't know when it would be useful. Maybe if you wrote code that used a timer to measure instruction  execution time?