dcwbrown/dwire-debug #13: Integrate amazing GDB mode from @kadamski

dcwbrown, 2017-01-28:
@kadamski has proposed a gdb supportmode and supplied an initial pull request.

Merge this code and adapt if necessary for cross platform support.

dcwbrown, 2017-02-01:
@kadamski's gdb server has been integrated to work on both Linux and Windows. 

@kadamski: I believe this now works as well ass your protoype, and has the following minor improvements:

 - The `info registers` command now picks up the PC correctly.
 - dwdebug gdbserver mode exits cleanly when quitting GDB.

Note that the gdbserver mode is now a dwdebug command rather than a special '-' parameter. 

This is to allow the port and baud rate to be selected before entering gdb server mode.

For example:

```
$ ./dwdebug device ttyUSB0 7840, gdbserver, q

Connected to DebugWIRE device on USB serial port COM4 at baud rate 7840
Device recognised as ATtiny85
Target ready, waiting for GDB connection.
Use 'target remote :4444'
```

---

Note that Windows sockets support uses a WinSock implementation which is slightly different in API from Linux. Apologies for a few grubby casts this causes. 

For example file handles need casting as (FileHandle) before passing to Read/Write.

kadamski, 2017-02-01:
That is cool, I'll take a look at this as soon as I can.

plasticassius, 2017-02-07:
I'm just starting to explore the possibilities here, and it looks like this is going to be quite useful. I still have to learn how to deal with complications like being able to single step without stepping into interrupt routines, but I imagine that's a matter of spending more time with gdb.

To avoid casting, what do you think of declaring handles as FileHandle even in the gdbserver directory? That might eliminate much of the casting (but it doesn't look like there's all that much to begin with). In fact, it looks quite clean to me.

dcwbrown, 2017-02-08:
Sounds fine to me. Although it may not look it I was trying not to change too much of @kadamski's code.

plasticassius, 2017-02-10:
I guess the important thing is that it's working well.

zoon81, 2017-03-19:
Hi.So I can use for example Eclipse for debugging my C code?

kadamski, 2017-03-19:
@zoon81: that should be possible. Why don't you give it a try and let us know? Watch out for breakpoints however. You don't want to use software breakpoints (which may be default for eclipse) but hardware breakpoint.

zoon81, 2017-03-19:
Thansk.I going to get an ft232 based converter and I give it a try.