dcwbrown/dwire-debug #17: Have quit start execution and device not reset

plasticassius, 2017-02-07:
As discussed in #15 , allow dwdebug to start and stop without affecting the microcontroller program.

By having quit start execution, dwdebug can be used as an in circuit loader. This is more convenient than an ISP programmer or bootloader since it works on a single pin.

By having device connect to the microcontroller without reseting it, dwdebug can be stopped and restarted at will without changing state. It is more convenient to not have to care about what order things are stopped and started in.

dcwbrown, 2017-02-08:
Much earlier on in development I was having trouble connecting and got worried that it may be possible for a running processor to lock out the debugwire hence the tendency to keep things stopped by default.

Now that it seems pretty stable, I thoroughly agree with you. I like the idea that the debugger shouldn't assume too much all by itself.

dcwbrown, 2017-02-08:
Aside: although there is no start address option on the go command, there is a 'p' command to set the pc, e.g.: 
```
dwdebug l my.elf, reset, p1800, g, q
```
I should add this to the docs.

dcwbrown, 2017-02-08:
Done, thanks.

plasticassius, 2017-02-10:
Thanks, this is great. I think that at this point it's quite useful. The p command will be handy also, looks like I overlooked it before.

plasticassius, 2017-02-12:
The first time the device is powered up, dwdebug fails:
```
dwdebug device,l build/tusb.elf ,reset,q
ttyUSB0 ...........
Connected to DebugWIRE device on USB serial port ttyUSB0 at baud rate 99988
Device recognised as ATtiny84
Loading 3152 flash bytes from ELF text segment 0 to addresses $0 through $c4f.
Unchanged   $0000 - $003f
... (more of the same)
Unchanged   $0c40 - $0c7f
Didn't receive 0x55 on reconnection, got 
92
Backtrace:
  dwdebug(DwSync+0x64) [0x405c72]
  dwdebug(DwReset+0x36) [0x4066ff]
  dwdebug(HandleCommand+0x99) [0x40d405]
  dwdebug(ParseAndHandleCommand+0x9b) [0x40d4d8]
  dwdebug(UI+0xc5) [0x40d6a3]
  dwdebug(main+0x2a) [0x40d703]
  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f8496265830]
  dwdebug(_start+0x29) [0x404149]

----------

dwdebug device,reset,l build/tusb.elf ,q
ttyUSB0 ...........
Connected to DebugWIRE device on USB serial port ttyUSB0 at baud rate 99988
Device recognised as ATtiny84
Didn't receive 0x55 on reconnection, got 
92
Backtrace:
  dwdebug(DwSync+0x64) [0x405c72]
  dwdebug(DwReset+0x36) [0x4066ff]
  dwdebug(HandleCommand+0x99) [0x40d405]
  dwdebug(ParseAndHandleCommand+0x9b) [0x40d4d8]
  dwdebug(UI+0xc5) [0x40d6a3]
  dwdebug(main+0x2a) [0x40d703]
  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f47186a5830]
  dwdebug(_start+0x29) [0x404149]
dwdebug device,reset,l build/tusb.elf ,q
ttyUSB0 ..................
Connected to DebugWIRE device on USB serial port ttyUSB0 at baud rate 62473
Device recognised as ATtiny84
Loading 3152 flash bytes from ELF text segment 0 to addresses $0 through $c4f.
Unchanged   $0000 - $003f
... (more of the same)
Unchanged   $0c40 - $0c7f
```
In both of the cases above, when I repeat the command, it works. And, it continues working from then on. It's the first attempt after power up that fails. While it's easy enough to work around this, it might be worth figuring out what's causing this. Do you have any ideas about what `Didn't receive 0x55 on reconnection, got 92` could mean?

dcwbrown, 2017-02-12:
0x55 is what a dwire device responds following break - chosen presumably for the convenience of the bit pattern 01010101. But you are seeing in this case 10010010 - which suggests somehow the clock has changed to something somewhat slower.

And indeed your logs above show an initial rate of 99,988 baud and a later rate of 62,473 baud.

I can only guess that the code that you are loading is changing the clock settings or divisor?

You may be able to work around this with a device command between the load and reset commands.

(BTW your initial device command is redundant - dwdebug automatically runs a device command whenever a command (such as load) that needs to be connected is run when the device is unconnected.)

Cheers -- Dave.

plasticassius, 2017-02-12:
Ahh! That makes sense, the clock doesn't get set up until the code runs the first time, so the first run starts with the clock rate as given by the fuse settings. After that, the clock is tuned up to 12.8 MHz where it stays from then on. Thanks for pointing out what was happening, I hope I get used to this baud / clock rate relationship soon. I'll be trying to apply dwdebug wherever I can.

Cheers, Peter

plasticassius, 2017-02-12:
Apparently, it's more complicated than I thought. It turns out that it is the reset command that makes it lose connection:
```
dwdebug reset,q
ttyUSB0 ...................
Connected to DebugWIRE device on USB serial port ttyUSB0 at baud rate 63129
Device recognised as ATtiny84
dwdebug reset,q
ttyUSB0 ...........
Connected to DebugWIRE device on USB serial port ttyUSB0 at baud rate 99988
Device recognised as ATtiny84
Didn't receive 0x55 on reconnection, got 
92
Backtrace:
  dwdebug(DwSync+0x64) [0x405c72]
  dwdebug(DwReset+0x36) [0x4066ff]
  dwdebug(HandleCommand+0x99) [0x40d405]
  dwdebug(ParseAndHandleCommand+0x9b) [0x40d4d8]
  dwdebug(UI+0xc5) [0x40d6a3]
  dwdebug(main+0x2a) [0x40d703]
  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7feb03c84830]
  dwdebug(_start+0x29) [0x404149]
dwdebug reset,q
ttyUSB0 ..................
Connected to DebugWIRE device on USB serial port ttyUSB0 at baud rate 62473
Device recognised as ATtiny84
dwdebug reset,q
ttyUSB0 ...........
Connected to DebugWIRE device on USB serial port ttyUSB0 at baud rate 99988
Device recognised as ATtiny84
Didn't receive 0x55 on reconnection, got 
92
Backtrace:
  dwdebug(DwSync+0x64) [0x405c72]
  dwdebug(DwReset+0x36) [0x4066ff]
  dwdebug(HandleCommand+0x99) [0x40d405]
  dwdebug(ParseAndHandleCommand+0x9b) [0x40d4d8]
  dwdebug(UI+0xc5) [0x40d6a3]
  dwdebug(main+0x2a) [0x40d703]
  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f73bb711830]
  dwdebug(_start+0x29) [0x404149]
```
It seems that when reset changes the clock rate, the device does not restart. But when it's run again, the clock rate is already at the default, thus doesn't change, and the device starts properly.

Running a device command after reset makes it fail every time:
```
dwdebug reset,device,q
ttyUSB0 ..................
Connected to DebugWIRE device on USB serial port ttyUSB0 at baud rate 62473
Device recognised as ATtiny84
Couldn't find a DebugWIRE device.
Backtrace:
  dwdebug(ConnectSerialPort+0x7e) [0x407380]
  dwdebug(DeviceCommand+0xca) [0x4091e8]
  dwdebug(HandleCommand+0x99) [0x40d405]
  dwdebug(ParseAndHandleCommand+0x9b) [0x40d4d8]
  dwdebug(UI+0xc5) [0x40d6a3]
  dwdebug(main+0x2a) [0x40d703]
  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7fcb6c897830]
  dwdebug(_start+0x29) [0x404149]
dwdebug reset,device,q
ttyUSB0 ..................
Connected to DebugWIRE device on USB serial port ttyUSB0 at baud rate 62473
Device recognised as ATtiny84
Couldn't find a DebugWIRE device.
Backtrace:
  dwdebug(ConnectSerialPort+0x7e) [0x407380]
  dwdebug(DeviceCommand+0xca) [0x4091e8]
  dwdebug(HandleCommand+0x99) [0x40d405]
  dwdebug(ParseAndHandleCommand+0x9b) [0x40d4d8]
  dwdebug(UI+0xc5) [0x40d6a3]
  dwdebug(main+0x2a) [0x40d703]
  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f7fd96c2830]
  dwdebug(_start+0x29) [0x404149]
```
Removing both reset and device from the command makes it work every time:
```
dwdebug l build/tusb.elf ,q
ttyUSB0 ...........
Connected to DebugWIRE device on USB serial port ttyUSB0 at baud rate 99988
Device recognised as ATtiny84
Loading 3140 flash bytes from ELF text segment 0 to addresses $0 through $c43.
Unchanged   $0000 - $003f
...
Unchanged   $0c40 - $0c7f
dwdebug l build/tusb.elf ,q
ttyUSB0 ...........
Connected to DebugWIRE device on USB serial port ttyUSB0 at baud rate 99988
Device recognised as ATtiny84
Loading 3140 flash bytes from ELF text segment 0 to addresses $0 through $c43.
Unchanged   $0000 - $003f
...
Unchanged   $0c40 - $0c7f
```
In fact, `dwdebug l build/tusb.elf ,q` works every time regardless of whether the elf file changes or not. This is the magic I was looking for.

closed 2017-02-17 by plasticassius