dcwbrown/dwire-debug #24: Supported USB-TTL adaptors

a-v-s, 2017-08-26:
I have been testing and I found that CH340 based adaptors appear to work as well.

dcwbrown, 2017-08-26:
Thank you André, that's a nice surprise :=).

Questions ...
- which OS was this on?
- were you using default OS drivers, and if not, what did you need to do to install some?

a-v-s, 2017-08-26:
I am using ArchLinux (kernel version 4.12.8-2-ARCH) and I am the driver the default driver (ch341).

dcwbrown, 2017-08-27:
Thanks André, I'll update the readme. I have also ordered a couple of CH340 usb to serial adapters from ebay and will (someday) try them on WIndows.

kadamski, 2017-09-10:
Works for me too, in case you would like separate confirmation. On Linux, of course, but different distro. Also default driver.

dcwbrown, 2017-09-11:
Excellent, and I can confirm that it is working on Windows 10. Nice to have an alternative to the FT232.

closed 2017-09-11 by dcwbrown