dcwbrown/dwire-debug #25: lwire version search for ports / devices

deiKruve, 2017-09-06:
Hallo,
I made a setup with an FT2232, which gives 2 ttyUSB ports.
ther result of ls -U is:
----------------------------
jan@coconut:/dev$ ls -U|grep USB
ttyUSB1
ttyUSB0
jan@coconut:/dev$ 
-----------------------------
Dwire is connected to ttyUSB0.
With the Master branch I have no problem connecting, but with the lwire branch it proves impossible since ttyUSB1 is seen first, even if I specify ttyUSB0 on the command line.

Could you please look into that, since the lwire version has some new commands I sorely miss in the master version.

Otherwise: thumbs up for your excellent idea. I did some work already with an ft232 serial port and the lwire version. it works like a dream. 
The 2 port idea is so i can have a regular rs232 connection to the test cpu.

By the way the cpu is attiny441.

Cheers,

jan.

dcwbrown, 2017-09-06:
Thanks Jan. Yes, there was indeed a problem specific to selecting serial port 0: I reserved port 0 to mean 'first available port' when working on the usbtiny code, and forgot to refactor when adding back serial support.
I have just committed a fix which I hope resolves this for you.

deiKruve, 2017-09-07:
Right
It now works (almost) as advertised. The 'ls' command is still as it was,
but that is no great deal as long as you describe that in the docs.

There is one other issue that might concern you:
I need to program the chip at least twice in most cases for a successful
programming cycle. But I think, but am not sure, that avr-dude had the same
problem.

Thanks a lot,

j.

dcwbrown, 2017-09-07:
Thanks. 

Re ls: I'm sorry, I missed the problem with the ls command? What was the issue there?

Re multiple programming: For me it works remarkably reliably, possibly because I'm using it out of circuit with the attiny connected only to the FT232 by very short wires. I did once try long wires (30cm or so) and it was very unreliable.

Cheers -- Dave.

deiKruve, 2017-09-08:
Hallo,
Here is an extract from the screen from start up:
----------------
root@coconut:/home/jan/git-snap!/project/src-tiny-test#
/home/jan/bin/dwdebug device ttyUSB0
Connected to ATtiny441 on /dev/ttyUSB0 at 61936 baud.
033e: 93cf  push  r28                   > reset
Didn't receive 0x55 on reconnection, got 80.
Clock speed may have changed, trying to re-sync.
..............0000: c01d  rjmp  003c (+30)> ls
/dev/ttyUSB1 .
----------------------

and there it will sit.
ttyUSB1 is connected to the my board via rs232.

Basically I think that when you are probing you might want to skip after a
certain time and proceed to the next channel.
I do realize though that this might interfere with the non-dwire channel's
communication.
But perhaps for  a first time user that is better than a hung system.

> Re multiple programming:
Ok my wires are 20+ cm long. This points to:
1. not enough bulk capacitance on the ft board (none on my board)
2. a timing problem.
3. perhaps a ringing problem.

The speed is low enough that normally it should work without  problem.

I do notice that the baudrate setting of the channel changes with the
divider in the CLKPR - Clock Prescale Register.

So presuming that 61936 baud actually was meant to be 57600 baud (at 8Meg
clock and no divider) I could work back from there to startup clock (1 meg)
and find that baudrate accurately, since that is where I do my programming.

Thanks for all your help.
I see that you are an Oberon fan. 👍

j.

dcwbrown, 2017-09-08:
Ah, I see, when dwdebug finds an FT232 it assumes there is an AVR attached doesn't give up trying to connect to it. Yes, I must fix that.

Re wiring - well it's a hack - by using a diode I'm relying on the pull up resistor in the dwire device, where a normal dwire programmer will drive the line high. Please would you try again with much shorter wires between the FT232 and the AVR? (Or use a digispark programmed with the usbtiny code provided as that does properly drive the debugwire line.)

Debugwire runs at clock/128 baud, so for a clock of exactly 8Mhz, that is 62,500 baud. The speed detection in ./dwdebug settles for the rate in the middle of the a bracket of working rates, so it will adapt to the variability of the internal clock generator which is affected by both voltage and temperature.

I used Pascal (CDC mainframe) and then Modula (excellent PC implementation by Jensen partners) for the first half of my career but then joining a large company had no choice but to use c/c++. I find C's foibles a constant low level irritation. With Oberon, Wirth has simplified once again. I do like it.

Cheers -- Dave.

dcwbrown, 2017-09-08:
OK, I've checked in a change to the TERMIOS2 timeout/byte count settings (which always scare me). 

It does seem to be working, and I believe that's because the changes I have already made for the usbtiny support enabled these settings.

Please let me know if it works for you.

Cheers -- Dave.

deiKruve, 2017-09-12:
Success!

root@coconut:/home/jan/git-snap!/project/src-tiny-test#
/home/jan/bin/dwdebug device ttyUSB0
Connected to ATtiny441 on /dev/ttyUSB0 at 7839 baud.
0010: c030  rjmp  0072 (+49)            > ls
/dev/ttyUSB1 .Expecting break byte 0x00, but no bytes read.
ATtiny441 on /dev/ttyUSB0 at 7839 baud.
Unconnected.                            > device ttyUSB0
Connected to ATtiny441 on /dev/ttyUSB0 at 7839 baud.
0010: c030  rjmp  0072 (+49)            >

Thanks,

j.

closed 2017-09-12 by dcwbrown