dcwbrown/dwire-debug #26: write data command

deiKruve, 2017-09-14:
Hallo,
can I zero a block of RAM with the 'wd' command?
I am short of memory so I would dearly like to see if I overwrite my tail with the stack.

Cheers
j.

dcwbrown, 2017-09-15:
It's not as easy as a  purpose built fill command, but you can provide up to 16 bytes at a time to wd, so
you can zero e.g. the 16 bytes at 0x90 with the command:

`wd90 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0`

How much do you need to fill, and is 16 bytes at a time workable for you? 

(If not, I should consider adding a set of fill commands.)

-- Dave.

deiKruve, 2017-09-15:
Hi,
my 2 cents is that 'wd' must stay as it is.

fil <address> <length> <fill-byte>
Would be excellent, but I can live without it, for the moment. If I fill
-say- 3 x 16 bytes that should tell me whether I have enough stackspace
left (out of 256 bytes).

I have been looking at the programming issue a bit closer:
I never need to write more than twice to program all locations properly.
Then I have increased the bulk capacitor on the supply to 10uF ceramic and
I put a big screencable-type ferrite around the connections to the fdti.
And I did look at your code.
It all made no difference at all. In fact I have never noticed any mangled
data on the connection.
So I guess it is the chip that does not program quite as good at 3.3 volt
supply.

Cheers,

j.