dcwbrown/dwire-debug #29: Programming does not work

a-v-s, 2017-10-19:
I am using a HL-340 USB-TTL adaptor. 
I can connect to a ATMEGA328P through dwdebug and debug it. However, writing to flash appears to fail.  I have tried both through gdb (using eclipse) and directly using the l command in dwdebug.It looks like it's running the old program code. When I quit debugwire and write the program to flash using avrdude, the new code can be debugged.

dcwbrown, 2017-10-20:
Thanks André, I really need to move the lwire branch back over master as it has 328 support fixed among other fixes and new features. I'll work on this now.

dcwbrown, 2017-10-20:
I've merged the lwire branch into master. 
Please would you `git pull`, `make clean`, and `make` again and let me know if it works for you?
Thanks -- Dave.

a-v-s, 2017-10-20:
Programming seems to work fine. Breakpoints are working as well. 
Thanks!

dcwbrown, 2017-10-20:
Excellent! Thanks!

closed 2017-10-20 by dcwbrown