dcwbrown/dwire-debug #31: 'td' command not working

FewerToWhom, 2017-11-24:
Greetings,

When playing with a simple code that uses time related interrupts to blink a LED,
I have noticed that the 'td' command does not seem to work. If I leave a break point
inside the interrupt handler, it is always called when I run the code, regardless of
calling 'td' beforehand.

Cheers.

dcwbrown, 2017-11-24:
The td command controls a flag which is used by the go command. Go will use a 0x4x debug wire command if timers are enabled, or a 0x6x command if they are disabled. 

According to the analysis at http://www.ruemohr.org/docs/debugwire.html this corresponds to the 'run timers' setting in the atmel debugger.

The Atmel docs describe this as follows:

>#### Keep timers running in stop mode
>When checked, the timers set in the program will continue to run even when the program breaks at breakpoint or is halted

So 'td' doesn't disable timers completely, they still run when the code is running.

FewerToWhom, 2017-11-24:
What is, then, the point of the 'td' and 'te' commands?

dcwbrown, 2017-11-24:
You would need to address that question to Atmel ...