dcwbrown/dwire-debug #34: Fix Linux build

nerdralph, 2018-03-29:
I'm a Linux/cli guy, and wanted to try out dwire-debug since the ReadMe indicates it will work with a ttl UART from the commandline.  From the readme:
"There are no library or include file prerequisites."

So far I removed -lusb from the Makefile, added a USE_USB ifdef around the USB and digispark source includes, and am still working on stripping out the UI code from the build.

ccrause, 2018-03-30:
Can you explain what your intention is by "stripping out the UI code"?  Do you mean you just want to start a gdbserver session and not use interactive debug commands?

dcwbrown, 2018-03-30:
Yes, please would you explain a little more what you are looking for?
There are no external dependency pre-requisites. 
There should be no need to strip anything out - what problem do you see when you try building it?
Cheers -- Dave.

nerdralph, 2018-03-30:
I wanted to try the CLI (src/commandline).  The first build error I
encountered was no libusb, so I removed -lusb from the build command.  Then
I got a bunch of errors about digiwire or digispark.

Yes, please would you explain a little more what you are looking for?
There are no external dependency pre-requisites.
There should be no need to strip anything out - what problem do you see
when you try building it?
Cheers -- Dave.

nerdralph, 2018-03-30:
Here's the exact build error from running 'make':
```
gcc -std=gnu99 -g -fno-pie -rdynamic -fPIC -Wall -o dwdebug src/dwdebug.c -lusb -ldl
In file included from src/system/system.c:2:0,
                 from src/dwdebug.c:3:
src/system/SystemUSB.c:92:19: fatal error: usb.h: No such file or directory
compilation terminated.
Makefile:48: recipe for target 'dwdebug' failed
make: *** [dwdebug] Error 1
```

dcwbrown, 2018-03-30:
I would expect usb.h to be part of your OS. On Ubuntu it's in /usr/include.
The make process assumes default include settings.
So two questions:
  What OS are you building on?
  Have you a non-default INCLUDE setting?
Thanks -- Dave.

dcwbrown, 2018-03-30:
(And I believe libusb should also be part of your default OS environment.)

nerdralph, 2018-03-30:
usb.h is not in /usr/include on Ubuntu (14 or 16).
If you have it on your Ubuntu install, then you likely have installed the package libusb-1.0-0-dev.
To find out what package a file came from, use 'dpkg -S', i.e.:
dpkg -S /usr/include/usb.h

It would be easy for you to catch these errors if you used a build/test service like Travis, CircleCI, or Codeship (all of which are free for open-source github projects).

dcwbrown, 2018-03-30:
Thanks. I have updated the readme to specify the dependency on libusb-dev.

Which of Travis, CircleCI, or Codeship do you use? do you have a preference?

nerdralph, 2018-03-30:
A couple years ago I picked Codeship, but I may take another look as most services now allow you to use your own docker images, so sudo rights in the build VM are not as important.
http://nerdralph.blogspot.ca/2015/04/continuous-integration-wonderful-world.html

nerdralph, 2018-03-30:
p.s. after installing libusb-dev, the build works.  Note that libusb-1.0-0-dev provides libusb.h, not usb.h.  "libusb-dev" is the older v0.1.12.

dcwbrown, 2018-03-30:
Thanks, I've added CI. 

After looking also at the github advice on CI, I ended up using Semaphore as a) it explicitly claims to support 'C' (others only say 'C++', and b) it comes first in github's list of CIs. It does seem to work OK.

Yes, I'm using the old v0.1 libusb support as v1.0 would require a lot more work and special-case code to provide cross platfrom (linux and windows) support.

Cheers -- Dave.