dcwbrown/dwire-debug #35: prefer resistor instead of diode

nerdralph, 2018-03-31:
I just finished testing dwdebug with a tiny13a running off the internal ~9.6Mhz oscillator.  The baud rate was ~72kbps, or a bit time of 13.9uS.  Using a 1n4148 diode connected to the TTL UART Tx line, rise time was ~3uS.  This is due to the weak pull-up (~45K Ohm) on the RST pin.
After doing some experimenting with resistors and diodes, I think a ~4.7K Ohm resistor between RST and Tx is the best idea.

nerdralph, 2018-04-21:
OK, after a lot more experiments, at least with a Pl2303, I found a 1.5K resistor is best:
http://nerdralph.blogspot.ca/2013/09/clone-pl-2303hx-usb-to-ttl-serial.html

RikusW, 2019-10-24:
I just tested this with a 2k2 resistor on a 1MHz MDIO bus that I accelerate using SPI instead of pure bitbanging. Seems to work, the pullup on the bus was 2k2 so I just moved it between MISO and MOSI.

RikusW, 2019-10-24:
http://nerdralph.blogspot.com/2018/04/debugging-debugwire.html