dcwbrown/dwire-debug #48: building on windows

gdecarp, 2021-05-06:
This tool looks exactly like what I was looking for. Thanks! Just tried it and i had some trouble with building today's master version (on Windows 10). I downloaded Cygwin, installed the mingw64-i686-binutils and mingw64-i686-gcc-core libraries, and run make. But the code has some easily-solved issues that the compiler and linker didn't like, so i thought i'd share my changes.
In Makefile, change line:
	i686-w64-mingw32-gcc -std=gnu99 -Wall -o $(BINARY) -Dwindows src/$(TARGET).c -lKernel32 -lowing -lComdlg32 -lWs2_32
to:
	i686-w64-mingw32-gcc -std=gnu99 -Wall -o $(BINARY) -Dwindows src/$(TARGET).c -lKernel32 -lwinmm -lComdlg32 -lWs2_32 
And in src/dwire/digispark.c, replace all 'bool' with 'int', all 'true' with '1', and all 'false' with '0'.
Cheers!

zoon81, 2021-05-07:
Cool, thanks gdecarp. Did you got it work? Last time, when I tried it, it did not worked. I tried with an Arduino Nano, ATmega328p, and now it is bricked, because I can not put back to ISP mode. Maybe my Arduino is a clone, I mean not assembled with an original Microchip part, I have no idea.

gdecarp, 2021-05-07:
Hi. I think i actually took DeqingSun branch, and that's the branch that needed these fixes, it seems. So dcwbrown's original code might not even need a fixup, at all! But yes, it works really well. I'm using an Arduino Nano as my USB<->COM bridge I had lying around (a clone, but with a proper FTDI USB chip, not the crappy CP210x chip), tying the Arduino's reset to ground (to prevent it from interfering), and using a single diode to combine the Nano's RX and TX pin to drive the DebugWire Reset pin on an ATTiny44 that I first burned to enable DebugWire. But I haven't tried to program a Nano this way. I read somewhere some Arduinos have a capacitor across their reset pin, which interferes with the intended use. Maybe yours has one as well?

gdecarp, 2021-05-07:
Btw, zoon81, the baudrate auto-detect didn't work for me, so i picked a manual value  (which is the CPU's speed divided by 128), and that worked just fine in my case. I hooked it up to Visual Studio (using makefiles), and use 'echo l myoutput.elf | dwdebug.exe device com3 62500' as the Debugging Command (for my 8Mhz target processor) when i hit F5.

zoon81, 2021-05-07:
@gdecarp I removed the cap from the reset line, and guess what, it is working now 🥳 
Thank you the tip man! I don't know why I didn't think about it. Have a nice day.
Cheers

gdecarp, 2021-05-07:
@dcwbrown All the code runs just out of the box on this original branch on Windows, in contrast to DeqingSun fork, so i'll close this issue because it shouldn't be here. At any rate, thanks for the project!

closed 2021-05-07 by gdecarp