dcwbrown/dwire-debug #51: Can you make it to support atmega328pb?

wanggaoteng, 2023-05-21:
I have tested, the dwdebug can not identify atmega328pb, can you make it to support this chip? 
atmega328pb use debugWIRE. 
Best regards.

dcwbrown, 2023-05-21:
Long time since I worked on this. I don't know how different the 328PB is from the 328P.
 
It may be sufficient to add the 328PB's signature to the characteristics table in src/dwire/connection.c:

The last four lines of the table are currently: 
  {"ATmega328P",  0x950F, 224, 2048, 1024, 32768, 0x31, 128, 0x3F00, 2, 0x1F, 0x22},
  {"ATmega328",   0x9514, 224, 2048, 1024, 32768, 0x31, 128, 0x3F00, 2, 0x1F, 0x22},
  {"ATmega32U2",  0x958A, 224, 1024, 1024, 32768, 0x31, 128, 0x0000, 0, 0x1F, 0x22},
  {0,                  0,   0,    0,    0,     0,    0,   0,      0, 0,    0}

You could try updating it by adding a 328PB line as follows:

  {"ATmega328P",  0x950F, 224, 2048, 1024, 32768, 0x31, 128, 0x3F00, 2, 0x1F, 0x22},
  {"ATmega328",   0x9514, 224, 2048, 1024, 32768, 0x31, 128, 0x3F00, 2, 0x1F, 0x22},
  {"ATmega328PB", 0x9516, 224, 2048, 1024, 32768, 0x31, 128, 0x3F00, 2, 0x1F, 0x22},
  {"ATmega32U2",  0x958A, 224, 1024, 1024, 32768, 0x31, 128, 0x0000, 0, 0x1F, 0x22},
  {0,                  0,   0,    0,    0,     0,    0,   0,      0, 0,    0}

(Based on the signature published in the 328PB datasheet section 32.3 table on page 388.)
 
Good luck -- Dave.

uffejakobsen, 2023-05-22:
Please post feedback here - if it works or not :-)

wanggaoteng, 2023-05-22:
> Long time since I worked on this. I don't know how different the 328PB is from the 328P.
>
> It may be sufficient to add the 328PB's signature to the characteristics table in src/dwire/connection.c: [...]

Hi, @dcwbrown 
Thank you for your great work, following your instruction, I add atmega328pb to the connection.c file, and recompile it. The dwdebug.exe works very well, it can identify atmega328pb now.
Best regards.

wanggaoteng, 2023-05-22:
> Please post feedback here - if it works or not :-)
Sorry, I'm late.
I just completed the test, it works very well, the dwdebug.exe is very useful.